amichai-bd/rvc_asap #5: Support the "EBREAK" instruction - make it End the test.

amichai-bd, 2021-11-02:
Support the "EBREAK" instruction - make it End the test.
This can be done by adding a simple output interface from the core.
The TB will check if the EBREAK signal asserted - in case it is it will $finish the test with a message

amichai-bd, 2021-11-30:
add An XMR (cross module reference) from the TB to look at the opcode.
Incase of a matchto "ebreke" start the "end_test" task.

amichai-bd, 2021-11-30:
You may use references from the lotr project 
https://github.com/amichai-bd/riscv-multi-core-lotr/tree/master/verif/tb

gilyaakov, 2021-12-04:
Greetings,

I added support for the ebrake command which runs the end_tb function.

In addition, I made an order in the apps folder.

The new folder includes the following subfolders:
asm, elf, C, sv, elf_txt for order and convenience.

I also started writing a script called buildl.sh which is currently compiling all the contents of the 's' folder into .elf files that are ejected into the 'elf' folder in 'apps'.

Later on the script will perform everything up to the level of test validity.

All of these changes are present in my branch push.

Have a wonderful week,
Gil.

gilyaakov, 2021-12-04:
Issue is closed.

closed 2021-12-04 by gilyaakov

amichai-bd, 2021-12-06:
@gilyaakov - I don't see the new branch with your changes.
Please make sure you pushed in correctly.

gilyaakov, 2021-12-06:
@amichai-bd 
It seems i have pushed it to our branch in mistake, That should be fine now (pushed it to the main).

amichai-bd, 2021-12-06:
@gilyaakov - seems like you are missing something in the way you are trying to push.
Can see it in this repository.
Please retry and make sure you see your changes in the repository. 
https://github.com/amichai-bd/rvc_asap/