amichai-bd/rvc_asap #51: Write the "Graphic screen library"

amichai-bd, 2022-06-06:
rvc_fwrite(int *ptr, int size, int *memOffset) //write to display a char array
rvc_printf ( const char * format, ... );

clearScreen()
drawPixel(int x, int y)
drawLine(int x1, int y1, int x2, int y2)
drawHLine(int x1, int x2, int y)
drawVLine(int x, int y1, int y2)
drawRectangle(int x1, int y1, int x2, int y2)
drawCircle(int cx, int cy, int r)

amichai-bd, 2022-06-22:
Make sure the lib is loadable - this way, it's not HW implementation-specific. (no need to add a new CR)
Simple Access to display as a memory-mapped IO device (MMIO).
You may have "ROM"/"DATA," part of the Graphic lib.

amichai-bd, 2022-08-07:
@gilyaakov  - I looks at the API you wrote, 
I see the declaration of the functions (draw_char) but not the functions itself  
[image]

Also in the "graphic_screen_test.c" you have all the definitions there- 
don't you need to "include" the API file?

gilyaakov, 2022-08-07:
All the API's files resides in library folder (in apps). The header files under inc folder and the implementations under src folder. The code that checks the API's is under examples. There is a Makefile for this.

run it by:

./buildl.sh -lib graphic_screen_api_test.c

amichai-bd, 2022-08-07:
Ok..
So why dos the "graphic_screen_test.c" have functions?
don't we need to "include" in the header?

gilyaakov, 2022-08-08:
It's just a sketch of the library, it's helps to check everything is ok. The library was build from this code. The code that linkage with libraries is in apps/library/examples