amichai-bd/rvc_asap #64: make a deticated test for all the possible lb,lh,lbu,lhu assemly.

amichai-bd, 2022-07-24:
clear:
    nop
    nop
    nop
    nop
    addi x1, x0, 0  
    addi x2, x0, 0  
    addi x3, x0, 0  
    addi x4, x0, 0  
    addi x5, x0, 0  
    addi x6, x0, 0  
    addi x7, x0, 0  
    addi x8, x0, 0  
    addi x9, x0, 0  
    addi x10, x0, 0  
    addi x11, x0, 0  
    addi x12, x0, 0  
    addi x13, x0, 0  
    addi x14, x0, 0  
    addi x15, x0, 0  
    addi x16, x1, 0  
    addi x17, x0, 0  
    addi x18, x0, 0  
    addi x19, x0, 0  
    addi x20, x0, 0  
    addi x21, x0, 0  
    addi x22, x0, 0  
    addi x23, x0, 0  
    addi x24, x0, 0  
    addi x25, x0, 0  
    addi x26, x0, 0  
    addi x27, x0, 0  
    addi x28, x0, 0  
    addi x29, x0, 0  
    addi x30, x0, 0  
    addi x31, x0, 0  
main:
  li    x1,  10
  li    x2,  0x04030201
  li    x3,  0xf0E0D0B0

  sw    x2  , 0(x1)
  lw    x10 , 0(x1)
  lh    x11 , 0(x1)
  lb    x12 , 0(x1)
  lhu   x13 , 0(x1)
  lbu   x14 , 0(x1)

  lh    x11 , 1(x1)
  lb    x12 , 1(x1)
  lhu   x13 , 1(x1)
  lbu   x14 , 1(x1)

  lh    x11 , 2(x1)
  lb    x12 , 2(x1)
  lhu   x13 , 2(x1)
  lbu   x14 , 2(x1)

  lh    x11 , 3(x1) # error
  lb    x12 , 3(x1) 
  lhu   x13 , 3(x1) # error
  lbu   x14 , 3(x1)

  li    x1,  0xdeadbeef
  li    x2,  20
  sb    x1,  0(x2) //0x000000ef
  sb    x1,  1(x2) //0x0000beef
  sb    x1,  2(x2) //0x00adbeef
  sb    x1,  3(x2) //0xdeadbeef

  li    x3,  30
  sh    x1,  0(x2) //0000beef
  sh    x1,  2(x2) //beefbeef

closed 2022-08-06 by gilyaakov

amichai-bd, 2022-08-06:
Did this test pass successfully?

gilyaakov, 2022-08-06:
Yes